petercorke/bdsim #2: Example of integrator of hybrid system on a Python project

traversaro, 2020-12-02:
Hi @petercorke, 
   thanks a lot for sharing such a tool, it seems to be quite interesting. 

I saw that in the README it is written that: 

> The biggest is that `bdsim` is based on a very standard variable-step integrator from the scipy library.  For discontinuous inputs (step, square wave, triangle wave, piecewise constant) the transitions get missed.  This also makes it inaccurate to simulate hybrid discrete-continuous time systems.  We really need a better integrator, perhaps [`odedc`](https://help.scilab.org/docs/6.1.0/en_US/odedc.html) from SciLab could be integrated.

Unfortunately I do not have a lot of time on working on this, but if anyone is interested in working on this, a good example of an integration of an advanced integrator that support for example hybrid system can be found in the [FMPy project](https://github.com/CATIA-Systems/FMPy). In particular, the `fmpy.sundials.cvode` contains a Python interface to the [Sundials' CVODE integrator](https://computing.llnl.gov/projects/sundials/cvode). An example of use of such interface for integrating a hybrid system (bouncing ball) can be found in https://github.com/CATIA-Systems/FMPy/blob/e1927a128a9552f20304587c54c716750e73b05f/tests/test_cvode.py .

petercorke, 2020-12-05:
Thanks for your interest, yours is the first comment!  

I'm happy to work with anybody who'd like to contribute something in this area.  I've got a few ideas sketched out about how I can use SciPy to do this better, but this project is on the back burner at the moment while I get the robotics and vision toolboxes up to a good level of maturity.

sawyerbfuller, 2021-08-18:
Dear Peter, great piece of software. We recently introduced a similar feature in python-control called [interconnect](https://python-control.readthedocs.io/en/0.9.0/generated/control.interconnect.html?highlight=interconnect) that uses a slightly different approach that entails naming signals. Your approach has the nice attribute that the code appears very simple. We will be keeping an eye on this! 

I read with some interest your remark that python is still in need of an integrator for hybrid discrete-continuous systems; we are definitely in need of such a feature for python-control. Thanks @traversaro for the pointer.

petercorke, 2021-08-19:
Hi @sawyerbfuller, thanks for contacting me.  I think I looked at interconnect a while back, and I think there’s some interesting things we could do together.  Quite complementary functionality.

Some students have been developing a GUI for me, see the attached.  The model is saved in JSON and I can build a bdsim model from that, and run it.  Still a work in progress quite useful I think.

I’ve added rudimentary hybrid system support which is ok for simple cases, but I’d like a stronger solution for that.

peter

sawyerbfuller, 2021-08-19:
Hi @petercorke looks like maybe github stripped the attachment? You can email it directly to me if easier at minster@uw.edu .

Very interested in your gui work, because not having a simulink-like gui -- and/or a hybrid integrator we can use with interconnect -- has been a main limitation for students using python-control vs matlab in the controls class I have been teaching.

petercorke, 2023-07-06:
The Scipy [solve_ivp](https://docs.scipy.org/doc/scipy/reference/generated/scipy.integrate.solve_ivp.html) integrator has event handling which might be usable.  Would need to add a `guard` method to each block that throws events and pass a list of these to the solver.

sawyerbfuller, 2023-08-14:
Looks like the docstrings in solve_ivp have been updated to clarify something that I had been confused by before: that setting t_eval seemed to have no effect on the points that were evaluated. Now it is clear that its purpose is only to specify which points are returned. 

That said, indeed it looks like events could be used to specify time points to make sure to evaluate. But, it seems like specifiying evenly-spaced events would go through a quite roundabout process in the integrator if I understand correctly. You would need to specify a function that has repeated zero-crossings - the obvious choice being perhaps a sinusoid that crossed zero at every `dt` of each discrete-time system? And then the integrator would go and do a slow, interative zero-crossing search for each of those, rather than just evaluating at points you already knew ahead of time. This seems slow but better than nothing.

petercorke, 2026-09-27:
Progress update on this one: bdsim's simulation core now has real zero-crossing event detection built on `solve_ivp` (`declare_crossing_event`/scheduled event intervals), which is the class of solution discussed here for hybrid discrete-continuous systems. It's not a general guard/event framework like FMPy's CVODE interface, but it's a solid step in that direction. Leaving this open as the tracking issue for further hybrid-solver work.

petercorke, 2026-11-15:
Thank you for your interest in bdsim, @traversaro and @sawyerbfuller. bdsim 1.5.0 now has event handling for discontinuous blocks: zero-crossing events built on `solve_ivp` for integrator limits, `CLIP`, `DEADBAND`, `SIGN`, `ABS` and `SWITCH`, so the simulation lands exactly on the discontinuity. That covers the original request, so I'm closing this. A more general guard/event framework would be a separate issue.

closed 2026-11-15 by petercorke

traversaro, 2026-11-15:
Great, thanks!