petercorke/bdsim #3: PID controller implementation

zam101, 2021-02-19:
i'm trying to simulate a block diagrame that contain a pid controller,and from what i know pid transfert fonction is 
[image]
as we can see this is not a proper transfert function (degree of the numerator > degree of the denominator) and from the documentation i found that this is not suported .so when i try to define  the pid in the code like this 
[image]
it throws this error 
[image]
i want to know if there is any solution or we cant implement a pid yet with this package ?
thank you.

petercorke, 2021-02-20:
Not yet, sorry. The usual trick is to make the d term s/(a+a) where a is another design parameter, set somewhere toward the open loop bandwidth of the plant

MartinGrignard, 2023-03-10:
Hi @petercorke,
First, thanks for the work done on this wonderful package!
I'm also trying to build a PID controller with it and, following your previous answer that recommends adding a [low-pass filter](https://en.wikipedia.org/wiki/Low-pass_filter) to the derivative term, the expression becomes
```math
H(s) = K_p + \frac{K_i}{s} + \frac{K_d s}{1 + \alpha s},
```
where $\alpha = \frac{1}{\omega_0}$ and $\omega_0$ is the cutoff frequency of the system.

Focusing solely on the derivative term, if I write
```python
derivator = bd.LTI_SISO([K_d, 0], [alpha, 1])
```
I get the exact same error as @zam101:
```error
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[29], line 1
----> 1 derivator = bd.LTI_SISO([k_d, 0], [alpha, 1])

File [.../bdsim/run_sim.py:806), in BDSim.blockdiagram..new_method..block_init_wrapper(self, *args, **kwargs)
    804 def block_init_wrapper(self, *args, **kwargs):
--> 806     block = cls(*args, bd=bd, **kwargs)  # call __init__ on the block
    807     return block

File [.../bdsim/blocks/transfers.py:332), in LTI_SISO.__init__(self, N, D, x0, **blockargs)
    330 if x0 is None:
    331     x0 = np.zeros((n,))
--> 332 assert nn <= n, "direct pass through is not supported"
    334 # convert to numpy arrays
    335 N = np.r_[np.zeros((len(D) - len(N),)), np.array(N)]

AssertionError: direct pass through is not supported
```
Do you have some hints on how I could actually make it work?

petercorke, 2023-03-12:
I've been thinking about this and it's much more complex than I initially thought.  My previous answer to @zam101 was hasty and not well thought through, it's more complex than I said then.  I tried to put some of this down on a [wiki page](https://github.com/petercorke/bdsim/wiki/PID-control).  My main focus with `bdsim` right now is getting it ready for the forthcoming book, and after that a derivative block and PID block are top of the feature list.

petercorke, 2023-07-09:
Hi @zam101 and @MartinGrignard, check out the thread in Issue #18. There is a new implementation of PID on the master branch version.

closed 2023-07-09 by petercorke