petercorke/bdsim #6: what am I missing?

mrrigidbody, 2022-01-19:
Hi, good job. this is a very useful software 
I'm trying to use PSO optimization algorithm along with your software, but in the first step, I can not run any example of your software and I have no idea why!
I installed all modules except spatialmath, because it can not be installed. I installed spatialmath-python instead that I think it is ok.
I downloaded the project, and want to run eg1.py in example folder and I get some errors. what is going wrong? can you help me please?
this is one of my errors:

Traceback (most recent call last):
  File "C:\Users\argha\Desktop\rasa\bdsim-master\bdsim-master\examples\eg1.py", line 8, in <module>
    sim.blocks()
AttributeError: 'BDSim' object has no attribute 'blocks'
[image]

thank you in advance...

petercorke, 2022-01-20:
you say "download", did you clone the repo?  The pip version is way out of date, but changes have been happening quickly recently and I've not been updating PyPi

petercorke, 2022-01-20:
"I installed all modules except spatialmath, because it can not be installed. I installed spatialmath-python instead that I think it is ok."  Am confused by this, not sure what you mean by "spatialmath" but "spatialmath-python" is what you need.

mrrigidbody, 2022-01-20:
thanks for the answer. 
I cloned the bdsim and run the eg1.py example again, but I've got another error. 
[image]

petercorke, 2022-01-20:
Ok, the fact I had 24 unpushed commits might be an issue. sorry.  do a git pull and see how it goes.

petercorke, 2022-02-12:
Did you get anywhere with this?

closed 2022-04-23 by petercorke