petercorke/bdsim #7: AttributeError during running the sample code

MerdanBay, 2022-01-29:
## Environment
. Windows 10 + conda+ python3.9
. __bdsim__ is installed by `pip install git+https://github.com/petercorke/bdsim.git`
. __machinevision__ is installed by `pip install git+https://github.com/petercorke/machinevision-toolbox-python.git`
. __roboticstoolbox__ is installed by `pip install git+https://github.com/petercorke/robotics-toolbox-python.git`

## ISSUE
When i try to run the sample code, i got these returns

```python
>>> import bdsim
>>> sim = bdsim.BDSim()
package bdsim not found
package roboticstoolbox not found
package machinevisiontoolbox not found
```
and then
```python
>>> bd = sim.blockdiagram()
>>> sim.blocks()
0  blocks loaded
>>> demand = bd.STEP(T=1, name='demand')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'BlockDiagram' object has no attribute 'STEP'
```
any idea why this happens?

petercorke, 2022-02-09:
Sorry, no idea.  I'm not using Windows.

It'd help if you could debug `load_blocks` which is toward the end of `bdsim.py` and is invoked when you instantiate `BDSim`.

petercorke, 2022-04-23:
Were you able to resolve this? I have have only recently been able to test under windows, the current version on GH works with windows

MerdanBay, 2022-04-24:
@petercorke No, not yet. However, it would be very helpful if we had a working installation document

petercorke, 2022-04-24:
You've installed what needs to be installed.  If you get a chance please install the latest version from GH and let me know how it goes.

closed 2022-04-24 by petercorke