petercorke/bdsim #8: Missing examples in PyPI package

PlushBeaver, 2022-08-06:
Hi, `bdedit` can't start when `bdsim` is installed from PyPI:
```
# apt-get -y install python3-pyqt5
# pip3 install bdsim==0.9.0
(snip, no errors)
$ bdedit
Traceback (most recent call last):
  File "/usr/local/bin/bdedit", line 5, in <module>
    from bdsim.bdedit import main
  File "/usr/local/lib/python3.8/dist-packages/bdsim/bdedit/__init__.py", line 14, in <module>
    from .interface import *
  File "/usr/local/lib/python3.8/dist-packages/bdsim/bdedit/interface.py", line 15, in <module>
    from bdsim.bdedit.block_importer import *
  File "/usr/local/lib/python3.8/dist-packages/bdsim/bdedit/block_importer.py", line 14, in <module>
    from examples import docstring_parser as parser
ModuleNotFoundError: No module named 'examples'
$ find /usr/local/lib/python3.8/dist-packages/bdsim/ -name '*examples*'
(nothing found)
```

Tobias-Fischer, 2022-10-06:
This is also blocking a conda-forge release.

petercorke, 2023-01-22:
ok, this was a bit of a hack, the info `bdedit` needs now comes from `bdsim` proper so this file is no longer required.  updated on master and pypi

closed 2023-01-24 by petercorke