petercorke/bdsim #18: PID cannot be instantiated

dgrunberg, 2023-07-03:
Running a simple block with just a PID controller, I get the error "PID object has no attribute 'bd'"", see below.
I believe the problem is that in the PID class, super() is not called until the end, yet the first line of code in the __init__ function is ```subsystem = self.bd.runtime.blockdiagram()```
at which point self.bd does not yet exist.

In trying to fix this, I am not sure what PID should be subclassed from, as Deriv uses SubsystemBlock, but PID uses SubSystem, and LTI_SS uses TransferBlock.

I could try to fix this with a hint as to which class it should be based on.  Also the PID documentation refers to 2 inputs (ref and plant output) but the code seems to have 1 input.

Thanks for what looks like a great package.  

```
Python sys.version='3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]'
bdsim version 1.1.1
* Simulate temperature control system
package roboticstoolbox not found
package machinevisiontoolbox not found
Traceback (most recent call last):
  File "/home/grunberg/algopower/./bug.py", line 17, in <module>
    controller = bd.PID(P=10, D=0, I=1, I_band=100)
  File "/home/grunberg/venv/blocks/lib/python3.10/site-packages/bdsim/run_sim.py", line 969, in block_init_wrapper
    block = cls(*args, bd=bd, **kwargs)  # call __init__ on the block
  File "/home/grunberg/venv/blocks/lib/python3.10/site-packages/bdsim/blocks/transfers.py", line 625, in __init__
    subsystem = self.bd.runtime.blockdiagram()
AttributeError: 'PID' object has no attribute 'bd'
```
The file is
```
#!/usr/bin/env python
'''
Simulate temperature control system
'''
import matplotlib.pyplot as plt
import sys
import numpy as np
import scipy.signal as signal
import bdsim

print(f'Python {sys.version=}')
print(f'bdsim version {bdsim.__version__}')
sim = bdsim.BDSim()
bd = sim.blockdiagram()

ref = bd.STEP(T=1, off=0, on=.3,name='ref')
controller = bd.PID(P=10, D=0, I=1, I_band=100)
scope = bd.SCOPE(styles=['k','r--'])

# connect

bd.connect(ref, controller,scope[0])
bd.connect(controller, scope[1])

bd.compile()
sim.report(bd)
out = sim.run(bd, 100)
print(out)
```

petercorke, 2023-07-06:
Thanks for your interest and post.  PID is a late addition to `bdsim` and not fully tested, sorry.  Let me have a look over the weekend and get back to you.  Your test example will be helpful.

dgrunberg, 2023-07-06:
Ok, thanks.  I have a fix for another issue involving 2 signals added together not working when going into a LTI_SS block.  Would you be interested in a pull request fix?
-Dan

petercorke, 2023-07-06:
Sure.  The issue is with the SUM or LTI_SS block?  The former I would have thought/hoped was pretty solid.PeterSent from my iPad

dgrunberg, 2023-07-07:
Actually the issue is with the INTEGRATOR (I thought it was the SUM), I will post an issue, and then a fix.
-Dan

On 7/6/23 5:51 PM, Peter Corke wrote:
Sure.  The issue is with the SUM or LTI_SS block?  The former I would have thought/hoped was pretty solid.PeterSent from my iPad

petercorke, 2023-07-09:
Hi @dgrunberg, thanks again for your interest in this.  Looking at the PID code I confess it's a mess and I can't even recall where I was going with this...  I've rewritten it, it's simpler now, and it now implements all the passed arguments.  It's checked in on master branch, along with `examples/pid.py`.  As per the original docstrings, but not the initial implementation, this block has two inputs: plant output, plant reference output.

I'd be grateful if you could have a play and let me know how it goes.

dgrunberg, 2023-07-09:
Yes, seems to work now, thanks!  Although it took some head scratching to figure out that the default is I_band=0.0, which means the INTEGRATOR never works.  I did a pull request for that change in case you want it.

Thanks,
Dan

On 7/9/23 1:05 AM, Peter Corke wrote:

Hi @dgrunberg<https://github.com/dgrunberg>, thanks again for your interest in this. Looking at the PID code I confess it's a mess and I can't even recall where I was going with this... I've rewritten it, it's simpler now, and it now implements all the passed arguments. It's checked in on master branch, along with examples/pid.py. As per the original docstrings, but not the initial implementation, this block has two inputs: plant output, plant reference output.

I'd be grateful if you could have a play and let me know how it goes.

closed 2023-07-11 by petercorke