petercorke/bdsim #19: Not being able to launch bdedit

Ahmed8395, 2023-07-06:
Hi @petercorke,
I raised this issue in a comment earlier which is that a cannot launch **bdedit** on my computer and every time I run the file `bdedit.py` I get a message saying (**ModuleNotFoundError: No module named 'PyQt5'**).

I have MacBook Pro (2017), and as you said **bdedit** is not necessary to execute the diagrams, but I have worked with Simulink before and I thought this would be an important tool to use especially because I don't have a strong background in coding as I am a mechanical engineer and so working with blocks might be more appropriate for me. 

This package is great, but if there were more examples and tutorials online on just how it install it and use it properly, it would be even better. Thanks for the hard work that has been done here for us to use such an important tool.

petercorke, 2023-07-06:
It would be helpful if you could use the `bdedit` bug report template, it asks you a bunch of questions that would help diagnose the problem.  Did you do
```
pip install PyQt5
```

Ahmed8395, 2023-07-07:
No I did not, but now that I did, the following message appears (**ImportError: No module named PyQt5.QtWidgets**)

The following describes the issue in the bug report format.

**Describe the bug:** Not being able to launch the bdedit editor when running the python file `bdedit.py`

**To Reproduce Steps to reproduce the behavior I have done the following steps:**
**Go to:** the (bdsim) folder
**Click on:** the (bin) folder, and then open the file `bdedit.py` using Visual Studio code, and running the file
**I get the error:** ImportError: No module named PyQt5.QtWidgets
**Expected behavior:** from what I understood, running the `bdedit.py` file should launch a blank page bdedit editor.

Screenshot:

[image]

**Operating environment:**

**OS:** [MacOS]
**Version:** [Big Sur version 11.6.4]
**Your version of PyQT:** PyQt5

The output of running [pip install PyQt5]:
```
Ahmeds-MBP-2:~ ahmed$ pip install PyQt5
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Requirement already satisfied: PyQt5 in /usr/local/lib/python3.9/site-packages (5.15.9)
Requirement already satisfied: PyQt5-sip<13,>=12.11 in /usr/local/lib/python3.9/site-packages (from PyQt5) (12.12.1)
Requirement already satisfied: PyQt5-Qt5>=5.15.2 in /usr/local/lib/python3.9/site-packages (from PyQt5) (5.15.2)
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Ahmeds-MBP-2:~ ahmed$ 
```

petercorke, 2023-07-07:
Ahmed, this is not a `bdedit`problem, it's an issue with your python, pip or PyQt5 install.  I really can't help with this, I'm using more modern MacOS and PyQt5 works fine for me. You are going to have to google the error and fix it.  If you are not using `conda` I would strongly recommend you do.

Ahmed8395, 2023-07-07:
That is OK, Thank you very much Peter. I'll certainly look into how this can be fixed.

petercorke, 2023-07-09:
Hi @Ahmed8395, I put some [notes on the wiki](https://github.com/petercorke/bdsim/wiki/getting-going-with-pyqt5) which includes a test example.

closed 2023-07-09 by petercorke