petercorke/bdsim #23: [bdedit Enhancement] Switch from PyQT5 to PySide6?

Gjacquenot, 2023-08-11:
`bdedit` relies on PyQt5, which has a GPL license, and that seem to [create problems with PyQT5](https://github.com/petercorke/bdsim/commit/db1b575c98a44f172c4cdcfe6eb593f3461e54db)

Here is the license for PyQT5: *PyQt5 is released under the GPL v3 license and under a commercial license that allows for the development of proprietary applications.*

`bdsim` has a MIT license, but including PyQT5 makes the application a GPL application, unless buying a commercila license.

An alternative would be to switch from PyQT5 to [PySide6](https://doc.qt.io/qtforpython-6/PySide6/QtWidgets/index.html)

The main notable difference between PyQT5 to PySide6 is licensing — with PyQt6 being available under a GPL or commercial license, and PySide6 under a LGPL license.

The PySide6 wheel is an alias to other two wheels [PySide6_Essentials](https://pypi.org/project/PySide6_Essentials) and [PySide6_Addons](https://pypi.org/project/PySide6_Addons), which contains a predefined list of Qt Modules.

Switching from PyQT to PySide is documented [here](https://www.pythonguis.com/faq/pyqt6-vs-pyside6/). Little more complicated than `sed -i s/PyQT5/pyside/g source.py` but can be done easily...

Moreover, [QT](https://doc.qt.io/qtforpython-6) recommend to use PySide6.

QTCreator can be used with PySide6.

petercorke, 2023-08-11:
I did not know that! I will investigate and fix, but am a bit backlogged at the moment.  Do you know if PySide is easier to install on Apple silicon?  PyQt5 was a pain.

Gjacquenot, 2023-08-11:
I have no experience with Apple Silicon, but some already had the problem and it seems to be fixed as indicated:

- @cmaureir wrote [here](https://forum.qt.io/post/698133) *PySide 6.2.2+ provides support for Apple Silicon, so you could simple run a `pip install pyside6` and get it.*
- There is a dedicated blog [here](https://betterprogramming.pub/build-a-python-gui-app-in-apple-m1-e16098112e61)

So it might be solved...

cmaureir, 2023-08-18:
Yes, we support Apple Silicon since 6.2.2+ we have a node in our CI that builds universal wheels which are uploaded to PyPI https://pypi.org/project/PySide6/#files

petercorke, 2023-08-19:
I've started this process, but this also involves an effective change from Qt5 to Qt6 which has non trivial changes in methods and enums.  Will keep you posted on progress

petercorke, 2026-09-27:
Thank you for your interest in bdsim and for the detailed writeup, @Gjacquenot! This has since been done — `bdedit` now depends on `PySide6` (see `pyproject.toml`), and the PyQt5/GPL licensing concern no longer applies. Closing this out.

closed 2026-09-27 by petercorke