petercorke/bdsim #24: bdedit function block

mars-seo, 2023-11-30:
Dear Peter,

I appreciate your work. I have a brief question using bdedit.

I tried to use a custom function in the bdedit. Is it possible or not?

From the code (bdsim), I can easily understand that I can create a specific function or lambda for the block.

However, I don't clearly understand how to use the function block from bdedit side.

petercorke, 2024-05-27:
You can specify a function name in the block, but that function must be in scope when the .bd file is executed.  You can do this with a MAIN block that references a Python program that defines or imports the functions that you need.  This is a bit complex and not as well documented as it should be.  There are examples of how this works [here](https://github.com/petercorke/RVC3-python/tree/main/RVC3/models) and useful detail in the README.md file.

petercorke, 2024-05-27:
What are you using instead?