petercorke/bdsim #25: [bdedit BUG] problem with arial font although it seems it is installed

vahidajalluian, 2024-02-27:
**Describe the bug**
Here this is the log I get on Ubuntu 22.04

$ bdedit

bdedit is beta code and prone to random crashing, save your work often

Traceback (most recent call last):
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface_scene.py", line 143, in determineFont
    self._system_font = ImageFont.truetype("arial.ttf", 14)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 861, in truetype
    return freetype(font)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 858, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 203, in __init__
    self.font = core.getfont(
OSError: cannot open resource

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gambler/pythonEnv/bin/bdedit", line 8, in <module>
    sys.exit(main())
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/bdedit.py", line 114, in main
    window = InterfaceWindow(screen_resolution, args.debug)
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface_manager.py", line 35, in __init__
    self.initUI(resolution, debug)
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface_manager.py", line 39, in initUI
    self.interface = Interface(resolution, debug, self)
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface.py", line 78, in __init__
    self.initUI(resolution, debug, parent)
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface.py", line 121, in initUI
    self.scene = Scene(resolution, self.layout, main_window)
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface_scene.py", line 88, in __init__
    self.determineFont()
  File "/home/gambler/pythonEnv/lib/python3.10/site-packages/bdsim/bdedit/interface_scene.py", line 145, in determineFont
    self._system_font = ImageFont.truetype("Arial.ttf", 14)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 861, in truetype
    return freetype(font)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 858, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 203, in __init__
    self.font = core.getfont(
OSError: cannot open resource

vahidajalluian, 2024-02-27:
sorry
solved
It seems simple font installation is not enough,
I have tried the code below and it solved my issue
sudo apt install ttf-mscorefonts-installer
So if you will delete this issue

petercorke, 2024-05-27:
I added this to the new FAQ page.  Thanks for suggesting a solution.

closed 2024-05-27 by petercorke