petercorke/bdsim #26: [bdsim BUG] Cannot run examples in Python 3.12

mhinkkan, 2024-07-19:
This project looks really interesting to me! 

I tried to run the example `eg1.py`, but I got the following error message:

> * Example of continuous-time system
> * Copyright (c) 2021- Peter Corke
> package bdsim not found
> package roboticstoolbox not found
> package machinevisiontoolbox not found
> Traceback (most recent call last):
>   File "/home/mhinkkan/GitHub/bdsim/examples/./eg1.py", line 13, in <module>
>     demand = bd.STEP(T=1, name="demand")
>              ^^^^^^^
> AttributeError: 'BlockDiagram' object has no attribute 'STEP'

My setup is as follows:

* Ubuntu 24.04 LTS
* Python 3.12.3

I tried to run the example both from the command line and from the VSCode (in both cases using the virtual environment), with same end result. The same problem appears with other examples as well.

mhinkkan, 2024-07-19:
Not sure if this helps, but we experienced somewhat similar problems in our project, after updating to Python 3.12. In our case, the reason was that `setuptools` is no longer a core `venv` dependency in version 3.12: 
 
https://docs.python.org/3/library/venv.html

We managed to solve the problem simply by changing the build backend from `setuptools` to `hatchling`, which seems to be nowadays the default option here:

https://packaging.python.org/en/latest/tutorials/packaging-projects/

However, it might be that the reason is something else in your case (or the structure of your project may not allow this kind of a change).

By the way, the link to the installation instructions in the `README.md` does not work, i.e., it might also be that I missed something in the installation stage. 

In any case, looking forward to trying bdsim!

petercorke, 2024-08-01:
thanks for pointing this out.  Definitely seems to be a 3.12 thing, my trusty old 3.9 conda environment still works fine.  bdsim is starting off ok but it has failed to find the standard block library, that should happen in line 9.  That's why it fails with the STEP block, it hasn't found it.  The line "package bdsim not found" is also telling.  I'll dig into what's going on.

petercorke, 2024-08-01:
Quick and dirty fix, comment out line 17 of bdsim/blocks/displays.py.  It's an unused import of a function that has been removed from Numpy2.  np2 is causing lots of headaches...  let me know how it goes.

mhinkkan, 2024-08-01:
Commenting out the line `from numpy.lib.shape_base import expand_dims` solves the problem, thanks!

KwentiN-ui, 2024-08-01:
If numpy 2 is the issue here, may I suggest changing the requirements inside the `pyproject.toml` to
```
dependencies = [
    "numpy<2",
    "scipy", 
    "matplotlib", 
    "spatialmath-python",
    "ansitable",
    "progress",
]
```
It would be great to fix this issue for real but this would help for the moment 👍

petercorke, 2024-08-01:
Numpy2 is the future, so I'd rather do the fix properly so people aren't required to have both version of numpy in their environment.  STEP is a dynamically added method which confuses the linter, not sure if there is a way to tell it about this.

petercorke, 2024-08-05:
Version 1.1.2 just pushed to PyPI.  Let me know if any issues.

closed 2024-08-05 by petercorke